BerkeleyAutomation/FogROS2 #15: Release into Humble

vmayoral, 2022-04-08:
*This ticket tracks the process of releasing FogROS2 core package into [ROS 2 Humble](https://docs.ros.org/en/rolling/Releases/Release-Humble-Hawksbill.html#release-timeline).  Dates are pretty tight, so we'll play by the ear and hope for the best*.

- [ ] prepare for release:
  - [ ] code review for release
    - [x] Update SECURITY document, remove guidelines since they didn't make it to the paper
    - [x] Remove duplicate code
    - [x] LICENSE for the meta-package
    - [x] Add CHANGELOG.rst to the two packages to release `fogros2` and `fogros2_launch`
    - [x] Re-structure `fogros2_launch` to look like a real package
    - [ ] Remove duplicated code in FogROS2 launch extensions `fogros2_launch` (ping @jeffi and @KeplerC, there's currently lots of duplicated code and we should only include the extensions)
    - [ ] code re-formatting (extract examples and h264 into separate packages, as agreed)
    - [ ] review dependencies for `fogros2` and `fogros2_launch` in package.xml files
    - [ ] re-validate functionality
    - [ ] release branches
    - [ ] release tags
- [ ] release
  - [ ] Create release repo 
  - [ ] Do [`bloom` release](https://wiki.ros.org/bloom/Tutorials/FirstTimeRelease) https://github.com/ros2-gbp/fogros2-release
    - [ ] `bloom` should PR `rosdistro`, if not, **manually create a PR** 
  - [ ] `rosdistro` team to review package and merge PR (1-2 days, 1 month to get apt installable)
  - [ ] Monitor build and release status on [repo page](http://repo.ros2.org/status_page/) and [buildfarm](https://build.ros2.org/)
  - [ ] Bloom for `humble`

vmayoral, 2022-04-08:
Started working on the release process in the `main` branch. Quite a bit of work left to be done refactoring code. Will look at this in the coming few days as time allows for it.

Will try to in parallel advance with requests for the release.

vmayoral, 2022-04-09:
Sent the request to create the team and the release repo https://github.com/ros2-gbp/ros2-gbp-github-org/pull/42. Need to wait now for approval. In parallel we should be fulfilling the items on "*prepare for release*" above.

vmayoral, 2022-04-10:
Encountering quite a few inconsistencies in naming across the source code. @jeffi and @KeplerC, are we going for `FogROS2` or for `FogROS 2`, or maybe instead (simpler) `fogros2`?

vmayoral, 2022-04-10:
Spent a few hours re-reviewing simple execution example of Cloud Nodes and I wasn't able to get https://github.com/BerkeleyAutomation/FogROS2/blob/main/fogros2_examples/launch/talker.launch.py running. This deserves further attention as highlighted in https://github.com/BerkeleyAutomation/FogROS2/issues/19.

As is, I'd say we should hold the Humble release until basic examples are stable enough.

vmayoral, 2022-04-12:
Update, we've got the repos now https://github.com/ros2-gbp/ros2-gbp-github-org/pull/42#event-6411912498. I can confirm access to https://github.com/ros2-gbp/fogros2-release. Once outstanding issues of code functionality are addressed, we should be able to submit a bloom release and then follow up with a rosdistro PR.

vmayoral, 2022-05-23:
Shall we close this folks?

KDharmarajanDev, 2022-10-17:
Release is successful, closing this issue.

closed 2022-10-17 by KDharmarajanDev