BerkeleyAutomation/FogROS2 #18: Connectity seems to be messed up while setting up VPN in Arm-based workstations

vmayoral, 2022-04-10:
**Describe the bug**
Connectity seems to be messed up while setting up VPN in Arm-based workstations (e.g. Apple M1-based processors)

**To Reproduce**

```bash
docker run -it --platform linux/amd64 --rm --net=host --cap-add=NET_ADMIN fogros2
root@docker-desktop:/home/root/fog_ws# source install/setup.bash  # source fogros2 workspace as an overlay
root@docker-desktop:/home/root/fog_ws# ros2 topic list
1649578524.307783 [0]       ros2: ddsi_udp_create_conn: set IP_MULTICAST_IF failed: Bad Parameter
[ERROR] [1649578524.313711792] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/node.c:261'

with this new error message:

  'rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/node.c:413'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1649578524.316559333] [rcl]: Failed to fini publisher for node: 1
error creating node: rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/node.c:413
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

vmayoral, 2022-04-10:
This shouldn't be a priority since the same behavior isn't observed in `x86_64` with Linux-native OS setups.

JasonLKelly, 2022-09-11:
Similar issues found directly in rmw_cyclonedds: https://github.com/ros2/rmw_cyclonedds/issues/273

SimeonOA, 2022-11-28:
Similar Issue now  being experienced on a Docker Container using osrf/ros2 Docker image with amd64 architecture running on a Mac.

DevinLeamy, 2023-02-15:
Hi. Has this been resolved?