BerkeleyAutomation/FogROS2 #19: Interoperability issues

vmayoral, 2022-04-10:
I'm unable to run https://github.com/BerkeleyAutomation/FogROS2/blob/main/fogros2_examples/launch/talker.launch.py successfully and obtain basic interoperability. Neither between edge-cloud, nor between edge-edge.

From my inspection, VPN works just fine but there's something wrong in the setup.

This deserved further attention and development time.

jeffi, 2022-04-13:
@vmayoral can you try again on the humble branch?  I've made a few changes and I'm now able to run the talker example without issue.  If you do run into an issue the ssh changes I made should make it easier to find the root cause.

vmayoral, 2022-04-13:
@jeffi, gave it a try again. Failed twice with `humble` branch:

| Attempt 1 :x: | Attempt 2 ❌ |
|----------|------------|
| [image] | [image] | 

Here're a few additional observations:

- 👍 there's definitely an improvement with the prompt, so @jeffi's changes definitely added good.
- I think we should polish the 101 experience (the talker launchfile I demonstrate above) before exposing this to others.
- `humble` branch is inconsistent from a ROS 2 perspective, it should upon `rolling` in here https://github.com/BerkeleyAutomation/FogROS2/blob/humble/Dockerfile#L1, not on top of Galactic (Humble is branching out Rolling). I believe @KDharmarajanDev took care of this in `main`.
- we're missing in `humble` branch many of the changes introduced in `main` (sorry I sent them there, I just didn't know that `humble` was the actual "good" one, since generally, development happens in `main` and then you branch things out when ready, as appropriate). I think it'd be nice to cherry-pick many of the changes introduced in main regarding the README polishing or just rebase things together.
- I think it'd be nice to *default* to whatever development repo we're sticking with (right now, default is `main`).

jeffi, 2022-04-13:
We branched `humble` since it will undergo many changes that are inconsistent with the paper.  We branched recently, so it should be straightforward to merge everything onto the `humble` branch.  Anything on main that is inconsistent with the paper, we may need to back out.  Sorry for not being clear about this earlier.

What are the steps you're taking to reproduce the issues you see?  Also are you using an EC2 instance near you?  Or `us-west-1`?

vmayoral, 2022-04-13:
Been using us-west-1. This is known to work in the past and I would expect
bigger latencies, but still interoperability.

Do you guys get issues with machines in Europe?

jeffi, 2022-04-13:
I think some DDS changes have occurred. So it might be the issue.  Can you try with a closer instance?  That’s best practice anyways.  We’ll try with EU servers too

KeplerC, 2022-04-13:
I just ran the humble branch with docker, and it works fine on my end
[image]

KeplerC, 2022-04-14:
I repeated the steps that @vmayoral mentioned. I had the same problem whenever I start fogros container the first time.

Then my theory is that if the VPN's wg0 interface exists, but /etc/wireguard/wg0 file does not exist(because the container is new), then the interface is not deleted and restarted. Then the setup does not succeed.

jeffi, 2022-04-16:
(Just getting the issue up to date from conversations over Slack and email)

The issue appears to stem from `ROS_DOMAIN_ID`.  When we set it on the robot, the setting is not transmitted to the cloud instance.  The result is that robot and cloud were talking on different port ranges, and thus never to each other.  The workaround for now is to leave `ROS_DOMAIN_ID` unset on the robot (to match the cloud).  The fix will be to match the  `ROS_DOMAIN_ID` setting on the cloud.

vmayoral, 2022-04-16:
Tried latest changes in `humble` in a clean, Docker-bases setup and failed ❌ :

<details><summary>Error dump</summary>

```bash
root@xilinx:/home/root/fog_ws# ros2 launch fogros2_examples talker.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-04-16-06-52-48-253448-xilinx-69
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [root]: init
using ROS workspace:  /home/root/fog_ws
creating EC2 instance
security group id is ['sg-0e92c69af4995c85c']
i-04e0451b9131a77c3
EC2 instance is created with ip address: 13.57.190.164
Hit:1 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal InRelease
Get:2 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:3 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
Get:4 http://repo.ros2.org/ubuntu/main focal InRelease [4,676 B]
Get:5 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:6 http://packages.ros.org/ros2/ubuntu focal InRelease [4,676 B]
Get:7 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [1,712 kB]
Get:8 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/main Translation-en [320 kB]
Get:9 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/main amd64 c-n-f Metadata [14.9 kB]
Get:10 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [917 kB]
Get:11 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/restricted Translation-en [131 kB]
Get:12 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/restricted amd64 c-n-f Metadata [528 B]
Get:13 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [918 kB]
Get:14 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/universe Translation-en [205 kB]
Get:15 http://repo.ros2.org/ubuntu/main focal/main arm64 Packages [987 kB]
Get:16 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/universe amd64 c-n-f Metadata [20.6 kB]
Get:17 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [24.4 kB]
Get:18 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/multiverse Translation-en [7,336 B]
Get:19 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 c-n-f Metadata [592 B]
Get:20 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [42.2 kB]
Get:21 http://repo.ros2.org/ubuntu/main focal/main amd64 Packages [1,012 kB]
Get:22 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-backports/main Translation-en [10.1 kB]
Get:23 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [22.7 kB]
Get:24 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-backports/universe Translation-en [15.5 kB]
Get:25 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-backports/universe amd64 c-n-f Metadata [808 B]
Get:26 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages [1,012 kB]
Get:27 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [1,385 kB]
Get:28 http://security.ubuntu.com/ubuntu focal-security/main Translation-en [240 kB]
Get:29 http://security.ubuntu.com/ubuntu focal-security/main amd64 c-n-f Metadata [10.0 kB]
Get:30 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [859 kB]
Get:31 http://security.ubuntu.com/ubuntu focal-security/restricted Translation-en [122 kB]
Get:32 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 c-n-f Metadata [532 B]
Get:33 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [698 kB]
Get:34 http://security.ubuntu.com/ubuntu focal-security/universe Translation-en [124 kB]
Get:35 http://security.ubuntu.com/ubuntu focal-security/universe amd64 c-n-f Metadata [14.3 kB]
Fetched 11.2 MB in 2s (4,975 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following packages will be upgraded:
  locales
1 upgraded, 0 newly installed, 0 to remove and 173 not upgraded.
Need to get 3,871 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://us-west-1.ec2.archive.ubuntu.com/ubuntu focal-updates/main amd64 locales all 2.31-0ubuntu9.7 [3,871 kB]
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
debconf: unable to initialize frontend: Readline
debconf: (This frontend requires a controlling tty.)
debconf: falling back to frontend: Teletype
dpkg-preconfigure: unable to re-open stdin:
Fetched 3,871 kB in 0s (15.4 MB/s)
(Reading database ... 172136 files and directories currently installed.)
Preparing to unpack .../locales_2.31-0ubuntu9.7_all.deb ...
[... 246 lines cut ...]
Requirement already satisfied: wgconfig in /usr/local/lib/python3.8/dist-packages (0.2.2)
Requirement already satisfied: boto3 in /usr/local/lib/python3.8/dist-packages (1.21.6)
Requirement already satisfied: paramiko in /usr/local/lib/python3.8/dist-packages (2.9.2)
Requirement already satisfied: scp in /usr/local/lib/python3.8/dist-packages (0.14.4)
Requirement already satisfied: botocore<1.25.0,>=1.24.6 in /usr/local/lib/python3.8/dist-packages (from boto3) (1.24.6)
Requirement already satisfied: s3transfer<0.6.0,>=0.5.0 in /usr/local/lib/python3.8/dist-packages (from boto3) (0.5.1)
[... cut ...]
Requirement already satisfied: cffi>=1.1 in /usr/local/lib/python3.8/dist-packages (from bcrypt>=3.1.3->paramiko) (1.15.0)
Requirement already satisfied: pycparser in /usr/local/lib/python3.8/dist-packages (from cffi>=1.1->bcrypt>=3.1.3->paramiko) (2.21)
/home/root fog_ws
removing old workspace
mv: cannot move 'fog_ws' to a subdirectory of itself, 'fog_ws/fog_ws'
successfully extracted new workspace
[ERROR] [launch]: Caught exception in launch (see debug for traceback): No module named 'fogros2.actions'
```

</details>

I'll be taking a few days off but will re-engage with the discussion next week!

mjd3, 2022-04-21:
Superseded by #36

closed 2022-04-21 by mjd3