BerkeleyAutomation/FogROS2 #20: Unattended-upgr interfering with apt-get process

jeffi, 2022-04-13:
**Describe the bug**
Occasionally `apt-get` fails when setting up a VM due to `unattended-upgr`

```
E: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 4750 (unattended-upgr)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
E: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 4750 (unattended-upgr)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
E: Could not get lock /var/lib/dpkg/lock-frontend. It is held by process 4750 (unattended-upgr)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
```

**To Reproduce**
Launch the talker example at the right time of day... (not easy to reproduce)

jeffi, 2022-04-22:
We can disable unattended-upgr using the steps here:
https://unix.stackexchange.com/a/470710/247044
The best solution might be just to make this part of the AMI (and AMI creation steps).

KeplerC, 2022-04-28:
It seems @KDharmarajanDev  has resolved this issue. Close it for now

commit c241223a (2022-04-26): Changed AMIs to ones that have unattended upgrades disabled. Fixes #20
closed 2022-04-28 by KeplerC