BerkeleyAutomation/FogROS2 #28: fog list fails if instance is not fully set up yet

mjd3, 2022-04-15:
**Describe the bug**
If the instance is still starting up, running `ros2 fog list` results in a `FileNotFoundError`.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `ros2 launch fogros2_examples talker.launch.py`
2. While this is starting the EC2 instance, run `ros2 fog list`
3. Errors with stack trace:
[image]

**Expected behavior**
Should output a list of current fogros2 instances.

KeplerC, 2022-04-15:
The current design is that we only flush the `info` file after the instance is launched, but we still create a directory for storing temporary files such as pem keys. 

Is there any better way of doing this? (I know I need to add error handling)

mjd3, 2022-04-15:
Yeah I think there are multiple options here: 1) just check if the info file is not available yet and return an abbreviated description for this instance, 2) remove any instances from `ros2 fog list` that are not completely set up yet. I'd probably lean towards (1) but as long as it doesn't crash it's probably fine.

jeffi, 2022-04-28:
The fix for #33 now reuses a single security group for all runs.  We can use it to filter EC2 instances with the with `fog list` and related CLI commands.

```
  instances = boto3_client.describe_instances(
    Filters=[{"Name":"instance.group-name", "Values":["FOGROS2_SECURITY_GROUP"]}])
```
We could also consider using other AWS/EC2 functionality like tags.

jeffi, 2022-04-28:
I'm working on a more robust version that queries AWS directly.

closed 2022-04-29 by jeffi