BerkeleyAutomation/FogROS2 #37: Change from using `/tmp/fogros` to using `~/.fogros2`

jeffi, 2022-04-28:
Currently temporary and persistent files are placed in `/tmp`. System processes might delete contents of `/tmp`, hindering their access.

Proposed solution: use `~/.fogros2` for all persistent files.

jeffi, 2022-04-28:
Fix checked in.  Also centralized location of path.  c57ea5477aec737a75176b67d8d2170cfbcb4608

closed 2022-04-28 by jeffi

KeplerC, 2022-04-28:
I actually thought about this and this is what we did in FogROS1. If we use `~/.fogros2`, it might be possible in some scenarios that relative path ("~") does not work. We should at least leave a mark here, in case there will be any bugs of relative/absolute path in the future.  Another way of doing this is to use COLCON workspace's path.

jeffi, 2022-04-28:
Under what scenarios would `~/.fogros2` not work?  We're using https://docs.python.org/3/library/os.path.html#os.path.expanduser for compatibility across OSes.

KeplerC, 2022-04-28:
Oh this should work. I thought you were using "~" directly.