BerkeleyAutomation/FogROS2 #38: Make `sudo` portion of launch happen as early as possible

jeffi, 2022-04-28:
**Is your feature request related to a problem? Please describe.**
It can take a while after launch before the user is prompted for a password to setup the VPN.  Ideally all sudo operations could happen right after launch so that the user can walk away and rely on it launching.

KeplerC, 2022-04-28:
We can leave a `sudo ls` at the beginning of the AWS class, but the code might not that clean.

jeffi, 2022-04-28:
Not a bad idea, but that assumes that `sudo` is set up to not ask for a password each time.

KeplerC, 2022-04-28:
At least when I am deploying it on NUC, it only prompts for the first sudo command. I am not sure how it behaves on other machines

jeffi, 2022-04-28:
Yeah, that's the default on Ubuntu, I think.  But other systems will have different settings.  https://askubuntu.com/a/636095

jeffi, 2022-04-28:
I'd also suggest that if we can remove the need for a sudo password prompt altogether, that might be the most ideal solution (assuming we can do so without sacrificing security).

nikhiljha, 2022-05-10:
The VPN start script is now moved up (only in main branch, do we want to also have this in the humble branch)? Passwordless sudo with an SSH key is also reasonably safe (I would argue just as safe as what we're doing now, provided the user executing the ROS payload is different from the user with passwordless sudo access).

We can ship a cloud-init config (cloud-init is distro and cloud-provider agnostic) when booting up the image for the first time with the SSH key and just rely on sudo to let us through with no password for all future executions.

SimeonOA, 2022-05-12:
Needs to be merged/moved into Humble Branch

mjd3, 2022-05-14:
@nikhiljha can you PR this for humble?

closed 2022-11-07 by SimeonOA