BerkeleyAutomation/FogROS2 #45: colcon build warnings on Ubuntu 22.04

jeffi, 2022-04-29:
**Describe the bug**
colcon generates warnings during builds on Ubuntu 22.04.  See screenshot.

**To Reproduce**
```
cd ~/fog_ws
colcon build --merge-install
```

**Expected behavior**
No warnings.

**Screenshots**
```
$ colcon build --merge-install
Starting >>> fogros2 
Starting >>> fogros2_examples
--- stderr: fogros2_examples                                                
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< fogros2_examples [2.28s]
--- stderr: fogros2                   
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< fogros2 [2.32s]

Summary: 2 packages finished [2.67s]
  2 packages had stderr output: fogros2 fogros2_examples
```

KeplerC, 2022-04-29:
Actually, if you build ros from source, this error occurs to most of the
packages. But i didn’t look into why this is the case

KeplerC, 2022-05-02:
https://answers.ros.org/question/396439/setuptoolsdeprecationwarning-setuppy-install-is-deprecated-use-build-and-pip-and-other-standards-based-tools/
It seems the solution is to downgrade python from 3.10 to 3.8. Should we close this issue since the issue is not on our side?

mjd3, 2022-05-02:
@KeplerC the python versioning only makes sense for Ubuntu 20.04 where Python 3.8 is the default install. However, in 22.04, Python 3.10 will be the default. Looks like colcon is tracking this issue already though: https://github.com/colcon/colcon-core/issues/454

KeplerC, 2022-05-02:
Close this issue as it is tracked in elsewhere.

closed 2022-05-02 by KeplerC

jeychandar, 2022-12-15:
colcon build --symlink-install
Starting >>> lio_sam 
[Processing: lio_sam]                             
[Processing: lio_sam]                                     
[Processing: lio_sam]                                       
[Processing: lio_sam]                                        
--- stderr: lio_sam                                        
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to openni will be disabled
** WARNING ** visualization features related to openni2 will be disabled
** WARNING ** apps features related to openni will be disabled
In file included from /usr/include/boost/preprocessor/tuple/elem.hpp:23,
                 from /usr/include/boost/preprocessor/arithmetic/add.hpp:21,
                 from /usr/include/boost/mpl/aux_/preprocessor/def_params_tail.hpp:66,
                 from /usr/include/boost/mpl/aux_/na_spec.hpp:28,
                 from /usr/include/boost/mpl/not.hpp:20,
                 from /usr/include/boost/mpl/assert.hpp:17,
                 from /usr/include/pcl-1.10/pcl/point_traits.h:48,
                 from /usr/include/pcl-1.10/pcl/point_cloud.h:50,
                 from /home/praveen/ros2_ws/src/LIO-SAM/include/lio_sam/utility.hpp:21,
                 from /home/praveen/ros2_ws/src/LIO-SAM/src/imageProjection.cpp:1:
/home/praveen/ros2_ws/src/LIO-SAM/src/imageProjection.cpp:14:6: warning: ‘using uint16_t = uint16_t’ is deprecated: use std::uint16_t instead of pcl::uint16_t [-Wdeprecated-declarations]
   14 |     (uint16_t, ring, ring) (float, time, time)
      |      ^~~~~~~~
In file included from /usr/include/pcl-1.10/pcl/PCLHeader.h:10,
                 from /usr/include/pcl-1.10/pcl/point_cloud.h:47,
                 from /home/praveen/ros2_ws/src/LIO-SAM/include/lio_sam/utility.hpp:21,
                 from /home/praveen/ros2_ws/src/LIO-SAM/src/imageProjection.cpp:1:
/usr/include/pcl-1.10/pcl/pcl_macros.h:94:9: note: declared here
   94 |   using uint16_t [[deprecated("use std::uint16_t instead of pcl::uint16_t")]] = std::uint16_t;
      |         ^~~~~~~~
In file included from /usr/include/boost/preprocessor/tuple/elem.hpp:23,
                 from /usr/include/boost/preprocessor/arithmetic/add.hpp:21,
                 from /usr/include/boost/mpl/aux_/preprocessor/def_params_tail.hpp:66,
                 from /usr/include/boost/mpl/aux_/na_spec.hpp:28,
                 from /usr/include/boost/mpl/not.hpp:20,
                 from /usr/include/boost/mpl/assert.hpp:17,
                 from /usr/include/pcl-1.10/pcl/point_traits.h:48,
                 from /usr/include/pcl-1.10/pcl/point_cloud.h:50,
                 from /home/praveen/ros2_ws/src/LIO-SAM/include/lio_sam/utility.hpp:21,
                 from /home/praveen/ros2_ws/src/LIO-SAM/src/imageProjection.cpp:1:
/home/praveen/ros2_ws/src/LIO-SAM/src/imageProjection.cpp:29:6: warning: ‘using uint32_t = uint32_t’ is deprecated: use std::uint32_t instead of pcl::uint32_t [-Wdeprecated-declarations]
   29 |     (uint32_t, t, t) (uint16_t, reflectivity, reflectivity)
      |      ^~~~~~~~
In file included from /usr/include/pcl-1.10/pcl/PCLHeader.h:10,
                 from /usr/include/pcl-1.10/pcl/point_cloud.h:47,
                 from /home/praveen/ros2_ws/src/LIO-SAM/include/lio_sam/utility.hpp:21,
                 from /home/praveen/ros2_ws/src/LIO-SAM/src/imageProjection.cpp:1:
/usr/include/pcl-1.10/pcl/pcl_macros.h:96:9: note: declared here
   96 |   using uint32_t [[deprecated("use std::uint32_t instead of pcl::uint32_t")]] = std::uint32_t;
      |         ^~~~~~~~
In file included from /usr/include/boost/preprocessor/tuple/elem.hpp:23,
                 from /usr/include/boost/preprocessor/arithmetic/add.hpp:21,
                 from /usr/include/boost/mpl/aux_/preprocessor/def_params_tail.hpp:66,
                 from /usr/include/boost/mpl/aux_/na_spec.hpp:28,
                 from /usr/include/boost/mpl/not.hpp:20,
                 from /usr/include/boost/mpl/assert.hpp:17,
                 from /usr/include/pcl-1.10/pcl/point_traits.h:48,
                 from /usr/include/pcl-1.10/pcl/point_cloud.h:50,
                 from /home/praveen/ros2_ws/src/LIO-SAM/include/lio_sam/utility.hpp:21,
                 from /home/praveen/ros2_ws/src/LIO-SAM/src/imageProjection.cpp:1:
/home/praveen/ros2_ws/src/LIO-SAM/src/imageProjection.cpp:29:23: warning: ‘using uint16_t = uint16_t’ is deprecated: use std::uint16_t instead of pcl::uint16_t [-Wdeprecated-declarations]
   29 |     (uint32_t, t, t) (uint16_t, reflectivity, reflectivity)
      |                       ^~~~~~~~
In file included from /usr/include/pcl-1.10/pcl/PCLHeader.h:10,
                 from /usr/include/pcl-1.10/pcl/point_cloud.h:47,
[... 45 lines cut ...]
In file included from /usr/include/boost/preprocessor/tuple/elem.hpp:23,
                 from /usr/include/boost/preprocessor/arithmetic/add.hpp:21,
                 from /usr/include/boost/mpl/aux_/preprocessor/def_params_tail.hpp:66,
                 from /usr/include/boost/mpl/aux_/na_spec.hpp:28,
                 from /usr/include/boost/mpl/not.hpp:20,
                 from /usr/include/boost/mpl/assert.hpp:17,
                 from /usr/include/pcl-1.10/pcl/point_traits.h:48,
[... cut ...]
                 from /home/praveen/ros2_ws/src/LIO-SAM/include/lio_sam/utility.hpp:21,
                 from /home/praveen/ros2_ws/src/LIO-SAM/src/imageProjection.cpp:1:
/usr/include/pcl-1.10/pcl/pcl_macros.h:96:9: note: declared here
   96 |   using uint32_t [[deprecated("use std::uint32_t instead of pcl::uint32_t")]] = std::uint32_t;
      |         ^~~~~~~~
---
Finished <<< lio_sam [2min 3s]
                          
Summary: 1 package finished [2min 4s]
  1 package had stderr output: lio_sam

jeychandar, 2022-12-15:
[image]

I am new to this ..can you help me out! Thanks in advance