BerkeleyAutomation/FogROS2 #50: ParamValidationError when `ros2 fog connect` is not given any arguments

mjd3, 2022-05-04:
**Describe the bug**
If a user mistakenly does not use the `-n` option for `ros2 fog connect`, they get a ParamValidationError.

**To Reproduce**
Steps to reproduce the behavior:
1. `docker run --rm -it --net=host --cap-add=NET_ADMIN fogros2`
2. `aws configure`
3. `ros2 launch fogros2_examples talker.aws.launch.py`
4. `ros2 fog connect`

**Expected behavior**
Should output an error message explaining usage of the `connect` cmd.

**Screenshots**
[image]

mjd3, 2022-05-04:
Note that nearly the same behavior occurs when using `ros2 fog delete` with no arguments.
[image]

jeffi, 2022-05-04:
Probably best to make this bug apply to all CLI verbs--they likely have similar issues.

PR #54 (2022-05-04): CLI error checking and interpretable error messaging
- Fixes #50 (removes `--name` option in favor of a required argument -- can be `all` for `delete` -- and shifts burden of validation to `argparse`)
- Fixes #51 (adds error handling to check if AWS config is setup already)
- Formatting of `verb` directory
closed 2022-05-04 by mjd3