BerkeleyAutomation/FogROS2 #52: Docker talker/listener example results in rmw_cyclonedds_cpp error

mjd3, 2022-05-04:
**Describe the bug**
When running the talker/listener cloud example in a docker container, rmw_cyclonedds_cpp fails to create a domain because it is looking in the wrong place for the configuration file.

**To Reproduce**
Steps to reproduce the behavior:
1. `docker run -it --rm --net=host --cap-add=NET_ADMIN fogros2`
2. `aws configure`
3. `ros2 launch fogros2_examples talker.aws.launch.py`
4. See error below

**Expected behavior**
Listener should start on the remote machine and communicate with the talker.

**Screenshots**
[image]

KeplerC, 2022-05-04:
This is because @jeffi updated install.md and not the main README.md. Should we keep two copies of the tutorial?

mjd3, 2022-05-04:
Ah actually it is because the ros_entrypoint.sh script is not up to date for the current Dockerfile with jammy. I can fix it now.

PR #53 (2022-05-04): Fix Docker bugs, make remote distro depend on local distro
- Fixes #52 by updating `ros_entrypoint.sh` to use the correct path for CYCLONEDDS_URI
- Fixes #48 by replacing `rolling` with a ros distro variable in `cloud_node.py`
- Cleans up the current Dockerfile by removing unnecessary installs
closed 2022-05-04 by mjd3