BerkeleyAutomation/FogROS2 #61: Unused files in fogros2?

mjd3, 2022-05-05:
**Describe the bug**
It seems like there might be some files that are not used in `fogros2` (namely, `fogros2.py`, but potentially others as well given that only some are imported in `__init__.py`). We should be sure that only the files used in the package are included.

mjd3, 2022-05-09:
@KeplerC is `fogros2/verb/image.py` used?

nikhiljha, 2022-05-12:
```
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/aws.py:94: unused method 'delete' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/aws.py:250: unused method 'add_docker_container' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/aws.py:267: unused class 'RemoteMachine' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/aws.py:280: unused method 'delete' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/aws.py:347: unused method 'get_ssh_key' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/aws.py:485: unused method 'delete' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/command/fog.py:24: unused class 'FogCommand' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/command/fog.py:25: unused method 'add_arguments' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/command_builder.py:29: unused method 'save' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/launch_description.py:116: unused class 'FogROSLaunchDescription' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/launch_description.py:151: unused method 'visit' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/launch_description.py:210: unused method 'get_launch_arguments' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/launch_description.py:284: unused attribute '_conditionally_included' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/launch_description.py:285: unused attribute '_conditionally_included' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/launch_description.py:406: unused property 'deprecated' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/scp.py:29: unused variable 'CRED' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/scp.py:30: unused variable 'CEND' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/verb/delete.py:30: unused class 'DeleteVerb' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/verb/delete.py:31: unused method 'add_arguments' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/verb/image.py:29: unused class 'ImageVerb' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/verb/image.py:30: unused method 'add_arguments' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/verb/list.py:25: unused class 'ListVerb' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/verb/list.py:26: unused method 'add_arguments' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/verb/ssh.py:29: unused class 'SSHVerb' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/fogros2/verb/ssh.py:30: unused method 'add_arguments' (60% confidence)
/Users/njha/Development/BerkeleyAutomation/FogROS2/fogros2/launch/cloud.launch.py:25: unused function 'generate_launch_description' (60% confidence)
```

mjd3, 2022-05-12:
Closed by #68

closed 2022-05-12 by mjd3