BerkeleyAutomation/FogROS2 #64: test_copyright for fogros2 and fogros2_examples

KeplerC, 2022-05-11:
**Describe the bug**
The existing repo does not pass `test_copyright` test in in colcon build and ros2 build farm. 

**To Reproduce**
```
colcon build 
colcon test
colcon test-result --verbose
```

**the current result**
```
gdpmobile11@gdpmobile11-NUC7PJYH:~/ros_ws$ colcon test-result --verbose
build/fogros2/pytest.xml: 1 test, 1 error, 0 failures, 0 skipped
- fogros2
  <<< error message
    collection failure
  >>>
build/fogros2_examples/pytest.xml: 3 tests, 0 errors, 3 failures, 0 skipped
- fogros2_examples.test.test_copyright test_copyright
  <<< failure message
    AssertionError: Found errors
    assert 1 == 0
  >>>
- fogros2_examples.test.test_flake8 test_flake8
  <<< failure message
    AssertionError: Found errors
    assert 1 == 0
  >>>
- fogros2_examples.test.test_pep257 test_pep257
  <<< failure message
    AssertionError: Found code style errors / warnings
    assert 1 == 0
  >>>

Summary: 4 tests, 1 error, 3 failures, 0 skipped
```

KeplerC, 2022-05-11:
For fogros2_examples: 
```
gdpmobile11@gdpmobile11-NUC7PJYH:~/ros_ws$ ament_copyright src/FogROS2/fogros2_examples/
src/FogROS2/fogros2_examples/fogros2_examples/__init__.py: could not find copyright notice
src/FogROS2/fogros2_examples/fogros2_examples/listener.py: could not find copyright notice
src/FogROS2/fogros2_examples/fogros2_examples/talker.py: could not find copyright notice
src/FogROS2/fogros2_examples/launch/gqcnn_docker.launch.py: could not find copyright notice
src/FogROS2/fogros2_examples/launch/talker.aws.launch.py: could not find copyright notice
src/FogROS2/fogros2_examples/launch/talker.local.launch.py: could not find copyright notice
src/FogROS2/fogros2_examples/launch/vslam.launch.py: could not find copyright notice
7 errors, checked 10 files
```

For fogros2: 
```
gdpmobile11@gdpmobile11-NUC7PJYH:~/ros_ws$ ament_copyright src/FogROS2/fogros2/
src/FogROS2/fogros2/fogros2/__init__.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/aws.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/cloud_node.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/command/fog.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/command_builder.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/dds_config_builder.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/fogros2.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/name_generator.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/scp.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/util.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/verb/delete.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/verb/image.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/verb/list.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/verb/ssh.py: could not find copyright notice
src/FogROS2/fogros2/fogros2/vpn.py: could not find copyright notice
src/FogROS2/fogros2/launch/cloud.launch.py: could not find copyright notice
16 errors, checked 22 files
```

mjd3, 2022-05-12:
Should be resolved by #74

closed 2022-05-12 by mjd3