BerkeleyAutomation/FogROS2 #65: test_flake8 for fogros2 and fogros2_examples

KeplerC, 2022-05-11:
**Describe the bug**
The existing repo does not pass `test_copyright` test in in colcon build and ros2 build farm. 

**To Reproduce**
```
colcon build 
colcon test
colcon test-result --verbose
```

**the current result**
```
gdpmobile11@gdpmobile11-NUC7PJYH:~/ros_ws$ colcon test-result --verbose
build/fogros2/pytest.xml: 1 test, 1 error, 0 failures, 0 skipped
- fogros2
  <<< error message
    collection failure
  >>>
build/fogros2_examples/pytest.xml: 3 tests, 0 errors, 3 failures, 0 skipped
- fogros2_examples.test.test_copyright test_copyright
  <<< failure message
    AssertionError: Found errors
    assert 1 == 0
  >>>
- fogros2_examples.test.test_flake8 test_flake8
  <<< failure message
    AssertionError: Found errors
    assert 1 == 0
  >>>
- fogros2_examples.test.test_pep257 test_pep257
  <<< failure message
    AssertionError: Found code style errors / warnings
    assert 1 == 0
  >>>

Summary: 4 tests, 1 error, 3 failures, 0 skipped
```

KeplerC, 2022-05-11:
for fogros2_examples: 
```
gdpmobile11@gdpmobile11-NUC7PJYH:~/ros_ws$ flake8 src/FogROS2/fogros2_examples/
src/FogROS2/fogros2_examples/setup.py:13:80: E501 line too long (96 > 79 characters)
src/FogROS2/fogros2_examples/fogros2_examples/listener.py:26:80: E501 line too long (97 > 79 characters)
src/FogROS2/fogros2_examples/fogros2_examples/listener.py:30:80: E501 line too long (107 > 79 characters)
src/FogROS2/fogros2_examples/launch/vslam.launch.py:20:80: E501 line too long (112 > 79 characters)
src/FogROS2/fogros2_examples/launch/vslam.launch.py:23:80: E501 line too long (399 > 79 characters)
src/FogROS2/fogros2_examples/launch/vslam.launch.py:26:80: E501 line too long (273 > 79 characters)
src/FogROS2/fogros2_examples/launch/vslam.launch.py:30:80: E501 line too long (90 > 79 characters)
src/FogROS2/fogros2_examples/launch/vslam.launch.py:32:80: E501 line too long (90 > 79 characters)
src/FogROS2/fogros2_examples/launch/talker.local.launch.py:19:80: E501 line too long (95 > 79 characters)
src/FogROS2/fogros2_examples/launch/talker.local.launch.py:23:80: E501 line too long (92 > 79 characters)
src/FogROS2/fogros2_examples/launch/talker.local.launch.py:24:80: E501 line too long (88 > 79 characters)
src/FogROS2/fogros2_examples/launch/talker.aws.launch.py:41:80: E501 line too long (91 > 79 characters)
src/FogROS2/fogros2_examples/launch/talker.aws.launch.py:43:80: E501 line too long (92 > 79 characters)
src/FogROS2/fogros2_examples/launch/talker.aws.launch.py:44:80: E501 line too long (111 > 79 characters)
src/FogROS2/fogros2_examples/launch/gqcnn_docker.launch.py:19:80: E501 line too long (112 > 79 characters)
src/FogROS2/fogros2_examples/launch/gqcnn_docker.launch.py:21:80: E501 line too long (301 > 79 characters)
src/FogROS2/fogros2_examples/launch/gqcnn_docker.launch.py:23:80: E501 line too long (249 > 79 characters)
src/FogROS2/fogros2_examples/launch/gqcnn_docker.launch.py:26:80: E501 line too long (90 > 79 characters)
src/FogROS2/fogros2_examples/launch/gqcnn_docker.launch.py:28:80: E501 line too long (90 > 79 characters)
```

KeplerC, 2022-05-11:
for fogros2: 
```
gdpmobile11@gdpmobile11-NUC7PJYH:~/ros_ws$ flake8 src/FogROS2/fogros2/
src/FogROS2/fogros2/setup.py:11:80: E501 line too long (84 > 79 characters)
src/FogROS2/fogros2/setup.py:22:80: E501 line too long (144 > 79 characters)
src/FogROS2/fogros2/launch/cloud.launch.py:21:80: E501 line too long (100 > 79 characters)
src/FogROS2/fogros2/fogros2/fogros2.py:23:80: E501 line too long (114 > 79 characters)
src/FogROS2/fogros2/fogros2/fogros2.py:32:80: E501 line too long (90 > 79 characters)
src/FogROS2/fogros2/fogros2/util.py:39:80: E501 line too long (111 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:38:80: E501 line too long (100 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:39:80: E501 line too long (92 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:40:1: W293 blank line contains whitespace
src/FogROS2/fogros2/fogros2/aws.py:109:80: E501 line too long (94 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:124:80: E501 line too long (138 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:127:80: E501 line too long (267 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:136:80: E501 line too long (86 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:156:80: E501 line too long (129 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:164:80: E501 line too long (159 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:167:80: E501 line too long (105 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:184:80: E501 line too long (86 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:187:80: E501 line too long (95 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:189:80: E501 line too long (132 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:201:80: E501 line too long (104 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:207:80: E501 line too long (96 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:218:80: E501 line too long (105 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:244:80: E501 line too long (108 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:256:80: E501 line too long (86 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:300:80: E501 line too long (108 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:312:80: E501 line too long (92 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:321:80: E501 line too long (107 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:332:80: E501 line too long (107 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:336:80: E501 line too long (98 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:356:80: E501 line too long (86 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:359:80: E501 line too long (88 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:360:80: E501 line too long (89 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:380:80: E501 line too long (106 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:393:80: E501 line too long (90 > 79 characters)
src/FogROS2/fogros2/fogros2/aws.py:407:80: E501 line too long (86 > 79 characters)
src/FogROS2/fogros2/fogros2/vpn.py:57:80: E501 line too long (82 > 79 characters)
src/FogROS2/fogros2/fogros2/vpn.py:59:80: E501 line too long (83 > 79 characters)
src/FogROS2/fogros2/fogros2/vpn.py:71:80: E501 line too long (88 > 79 characters)
src/FogROS2/fogros2/fogros2/vpn.py:72:80: E501 line too long (101 > 79 characters)
src/FogROS2/fogros2/fogros2/vpn.py:73:80: E501 line too long (99 > 79 characters)
src/FogROS2/fogros2/fogros2/vpn.py:74:80: E501 line too long (98 > 79 characters)
src/FogROS2/fogros2/fogros2/dds_config_builder.py:37:80: E501 line too long (118 > 79 characters)
src/FogROS2/fogros2/fogros2/dds_config_builder.py:43:1: W191 indentation contains tabs
src/FogROS2/fogros2/fogros2/dds_config_builder.py:43:1: E101 indentation contains mixed spaces and tabs
src/FogROS2/fogros2/fogros2/dds_config_builder.py:44:1: E101 indentation contains mixed spaces and tabs
src/FogROS2/fogros2/fogros2/dds_config_builder.py:55:80: E501 line too long (231 > 79 characters)
src/FogROS2/fogros2/fogros2/dds_config_builder.py:59:20: W291 trailing whitespace
src/FogROS2/fogros2/fogros2/launch_description.py:26:80: E501 line too long (96 > 79 characters)
src/FogROS2/fogros2/fogros2/launch_description.py:28:1: E402 module level import not at top of file
src/FogROS2/fogros2/fogros2/launch_description.py:29:1: E402 module level import not at top of file
src/FogROS2/fogros2/fogros2/launch_description.py:30:1: E402 module level import not at top of file
src/FogROS2/fogros2/fogros2/launch_description.py:31:1: E402 module level import not at top of file
src/FogROS2/fogros2/fogros2/launch_description.py:32:1: E402 module level import not at top of file
src/FogROS2/fogros2/fogros2/launch_description.py:34:1: E402 module level import not at top of file
src/FogROS2/fogros2/fogros2/launch_description.py:35:1: E402 module level import not at top of file
src/FogROS2/fogros2/fogros2/launch_description.py:76:80: E501 line too long (82 > 79 characters)
src/FogROS2/fogros2/fogros2/launch_description.py:78:80: E501 line too long (83 > 79 characters)
src/FogROS2/fogros2/fogros2/launch_description.py:133:80: E501 line too long (90 > 79 characters)
src/FogROS2/fogros2/fogros2/launch_description.py:143:80: E501 line too long (87 > 79 characters)
src/FogROS2/fogros2/fogros2/launch_description.py:144:80: E501 line too long (86 > 79 characters)
src/FogROS2/fogros2/fogros2/launch_description.py:154:80: E501 line too long (93 > 79 characters)
[... cut ...]
src/FogROS2/fogros2/fogros2/verb/delete.py:60:80: E501 line too long (84 > 79 characters)
src/FogROS2/fogros2/fogros2/verb/delete.py:80:80: E501 line too long (100 > 79 characters)
src/FogROS2/fogros2/fogros2/verb/delete.py:82:13: F401 'concurrent.futures.as_completed' imported but unused
src/FogROS2/fogros2/fogros2/verb/delete.py:85:80: E501 line too long (88 > 79 characters)
src/FogROS2/fogros2/fogros2/verb/delete.py:86:80: E501 line too long (103 > 79 characters)
src/FogROS2/fogros2/fogros2/command/fog.py:8:80: E501 line too long (91 > 79 characters)
```

KeplerC, 2022-05-11:
fogros2_examples is fixed by commit [fe65c81](https://github.com/BerkeleyAutomation/FogROS2/commit/fe65c81aedd4f85243bf6c057e5bb5a88d341a3a)

mjd3, 2022-05-12:
This is fixed by #71

closed 2022-05-12 by mjd3