BerkeleyAutomation/FogROS2 #66: test_pep257 for fogros2 and fogros2_examples

KeplerC, 2022-05-11:
**Describe the bug**
The existing repo does not pass `test_pep257` test in in colcon build and ros2 build farm. 

**To Reproduce**
```
colcon build 
colcon test
colcon test-result --verbose
```

**the current result**
```
gdpmobile11@gdpmobile11-NUC7PJYH:~/ros_ws$ colcon test-result --verbose
build/fogros2/pytest.xml: 1 test, 1 error, 0 failures, 0 skipped
- fogros2
  <<< error message
    collection failure
  >>>
build/fogros2_examples/pytest.xml: 3 tests, 0 errors, 3 failures, 0 skipped
- fogros2_examples.test.test_copyright test_copyright
  <<< failure message
    AssertionError: Found errors
    assert 1 == 0
  >>>
- fogros2_examples.test.test_flake8 test_flake8
  <<< failure message
    AssertionError: Found errors
    assert 1 == 0
  >>>
- fogros2_examples.test.test_pep257 test_pep257
  <<< failure message
    AssertionError: Found code style errors / warnings
    assert 1 == 0
  >>>

Summary: 4 tests, 1 error, 3 failures, 0 skipped
```

KeplerC, 2022-05-11:
For fogros2_examples: 
```
gdpmobile11@gdpmobile11-NUC7PJYH:~/ros_ws$ pep257 src/FogROS2/fogros2_examples/
src/FogROS2/fogros2_examples/setup.py:1 at module level:
        D100: Missing docstring in public module
src/FogROS2/fogros2_examples/fogros2_examples/talker.py:1 at module level:
        D100: Missing docstring in public module
src/FogROS2/fogros2_examples/fogros2_examples/talker.py:19 in public function `main`:
        D103: Missing docstring in public function
src/FogROS2/fogros2_examples/fogros2_examples/listener.py:1 at module level:
        D100: Missing docstring in public module
src/FogROS2/fogros2_examples/fogros2_examples/listener.py:20 in public class `MinimalSubscriber`:
        D101: Missing docstring in public class
src/FogROS2/fogros2_examples/fogros2_examples/listener.py:21 in public method `__init__`:
        D102: Missing docstring in public method
src/FogROS2/fogros2_examples/fogros2_examples/listener.py:29 in public method `listener_callback`:
        D102: Missing docstring in public method
src/FogROS2/fogros2_examples/fogros2_examples/listener.py:33 in public function `main`:
        D103: Missing docstring in public function
src/FogROS2/fogros2_examples/fogros2_examples/__init__.py:1 at module level:
        D104: Missing docstring in public package
src/FogROS2/fogros2_examples/launch/vslam.launch.py:1 at module level:
        D100: Missing docstring in public module
src/FogROS2/fogros2_examples/launch/vslam.launch.py:18 in public function `generate_launch_description`:
        D103: Missing docstring in public function
src/FogROS2/fogros2_examples/launch/talker.local.launch.py:1 at module level:
        D100: Missing docstring in public module
src/FogROS2/fogros2_examples/launch/talker.local.launch.py:17 in public function `generate_launch_description`:
        D200: One-line docstring should fit on one line with quotes (found 3)
src/FogROS2/fogros2_examples/launch/talker.local.launch.py:17 in public function `generate_launch_description`:
        D400: First line should end with a period (not ')')
src/FogROS2/fogros2_examples/launch/talker.aws.launch.py:1 at module level:
        D100: Missing docstring in public module
src/FogROS2/fogros2_examples/launch/talker.aws.launch.py:18 in public function `ami_image`:
        D103: Missing docstring in public function
src/FogROS2/fogros2_examples/launch/talker.aws.launch.py:36 in public function `generate_launch_description`:
        D200: One-line docstring should fit on one line with quotes (found 3)
src/FogROS2/fogros2_examples/launch/gqcnn_docker.launch.py:1 at module level:
        D100: Missing docstring in public module
src/FogROS2/fogros2_examples/launch/gqcnn_docker.launch.py:17 in public function `generate_launch_description`:
        D103: Missing docstring in public function
```

mjd3, 2022-05-12:
Should be resolved by #74

closed 2022-05-12 by mjd3