BerkeleyAutomation/FogROS2 #113: H264 plugin not found

mgualti, 2023-01-25:
When setting some image topics to use h264, like this:

```
stream_topics = [
            ("/my_topic/input_image", "h264"),
            ("/my_topic/output_image", "h264")

```

I see the following error after running the launch file and before the workspace on the cloud is built:

```
[republish-2] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[republish-2]   what():  According to the loaded plugin descriptions the class image_transport/h264_pub with base class type image_transport::PublisherPlugin does not exist. Declared types are  image_transport/compressedDepth_pub image_transport/compressed_pub image_transport/raw_pub image_transport/theora_pub
[republish-3] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[republish-3]   what():  According to the loaded plugin descriptions the class image_transport/h264_pub with base class type image_transport::PublisherPlugin does not exist. Declared types are  image_transport/compressedDepth_pub image_transport/compressed_pub image_transport/raw_pub image_transport/theora_pub
[ERROR] [republish-2]: process has died [pid 91878, exit code -6, cmd '/opt/ros/rolling/lib/image_transport/republish raw h264 --ros-args -r __node:=republish_node2 -r in:=/my_topic/input_image -r out/h264:=/my_topic/output_image/h264'].
```

The local and cloud nodes continue to launch, but no messages are received on the cloud side for the topic, /my_topic/input_image/cloud; although the topic is visible from ```rostopic list```.

I followed the steps [here ](https://github.com/BerkeleyAutomation/FogROS2) and [here](https://berkeleyautomation.github.io/FogROS2/video_compression) to install the image transport plugins, and the h264_image_transport package is installed in fog_ws.

**Desktop**
 - OS: Ubuntu 20.04
 - ROS: Rolling
 - Cloud: AWS

KDharmarajanDev, 2023-01-30:
Hi, sorry for the late response. It seems that the H.264 encoder is missing. We removed the H.264 encoder from the package to reduce the dependencies when we were releasing this as a Humble package. Please include [this](https://github.com/BerkeleyAutomation/h264_image_transport) repo as a package in your workspace, and if you clone it recursively, it should contain both the necessary encoder and decoder. I will make changes in the documentation to reflect this, and will also try it out myself to make sure things are working.

mgualti, 2023-02-13:
It's working now, thanks!

closed 2023-02-13 by mgualti