BerkeleyAutomation/FogROS2 #115: Error reading SSH protocol banner in aws instance and scp: /home/ubuntu//ros_workspace.tar: Permission denied in gcp

new-c0der-1, 2023-04-29:
**Describe the bug**
I am trying to build and run this project. I have used Amazon Cloud and also Google Cloud but always getting error.

**To Reproduce**
In AWS
1. Create AWS instance
2. Copy ami and region info

In Google Cloud
1. Create project
2. Build project and run

Also build docker images but still getting errors.

**Expected behavior**
It should build, connect and run.

Amazon cloud in native
```
[WARN] [1682773686.022252877] [fogros2.scp]: Error reading SSH protocol banner[Errno 104] Connection reset by peer, retrying...
Exception (client): Error reading SSH protocol banner[Errno 104] Connection reset by peer
Traceback (most recent call last):
  File "/home/c0der/.local/lib/python3.10/site-packages/paramiko/transport.py", line 2268, in _check_banner
    buf = self.packetizer.readline(timeout)
  File "/home/c0der/.local/lib/python3.10/site-packages/paramiko/packet.py", line 374, in readline
    buf += self._read_timeout(timeout)
  File "/home/c0der/.local/lib/python3.10/site-packages/paramiko/packet.py", line 601, in _read_timeout
    x = self.__socket.recv(128)
ConnectionResetError: [Errno 104] Connection reset by peer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/c0der/.local/lib/python3.10/site-packages/paramiko/transport.py", line 2091, in run
    self._check_banner()
  File "/home/c0der/.local/lib/python3.10/site-packages/paramiko/transport.py", line 2272, in _check_banner
    raise SSHException(
paramiko.ssh_exception.SSHException: Error reading SSH protocol banner[Errno 104] Connection reset by peer
```

Google cloud in docker log
```
Installing collected packages: scp
Successfully installed scp-0.14.5
Defaulting to user installation because normal site-packages is not writeable
Collecting wgconfig
  Downloading wgconfig-0.3.0-py3-none-any.whl (20 kB)
Installing collected packages: wgconfig
Successfully installed wgconfig-0.3.0
/home/root fog_ws
removing old workspace
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: scp: /home/ubuntu//ros_workspace.tar: Permission denied

root@DESKTOP-KR8NT80:/home/root/fog_ws#
```

KeplerC, 2023-05-01:
Hi Thanks for your interest.  My hypothesis is the configuration of firewall/security group that resets the ssh session. Can you try to directly ssh to the machine (get the IP address from the cloud console, with the ssh key in `~/.fogros/YOUR_INSTANCE_NAME.pem`)?

kryptoniancode, 2023-05-03:
I am able to connect to google cloud after changin username in fogros2 similar to local ubuntu username. May folder permission issues. 
For AWS the vm stopped immediatly after creation.

Also ros2 fog list does not show google cloud vms.

new-c0der-1, 2023-05-03:
I am able to connect to google cloud after changing username in fogros2 similar to local ubuntu username. May folder permission issues. 
For AWS the vm stopped immediatly after creation.

Also ros2 fog list does not show google cloud vms.

new-c0der-1, 2023-05-03:
I don't why this is the issues but this solved for me.

```python
def work_dir():
    global _work_dir_cache
    if _work_dir_cache is None:
        # home = os.path.expanduser("/home/new-c0der/")
        path = os.path.join("/home/new-c0der/.fogros2")
        _mkdir(path)
        _work_dir_cache = path
    return _work_dir_cache
```